square/pylink #179: AttributeError: module 'pylink' has no attribute 'JLink'.

s-bruce13, 2023-06-14:
I upgraded to latest version 1.1.0, but i get the following error:
AttributeError: module 'pylink' has no attribute 'JLink'.

I reversed back to the version i had before 0.8.2 and it works ok.

hkpeprah, 2023-06-14:
I don't believe we added any new dependencies. What do you see if you run the file directly using the interpreter?

s-bruce13, 2023-06-14:
What do you mean by interpreter?

hkpeprah, 2023-06-14:
Something like this:
```
$ python -m pylink.jlink
```

s-bruce13, 2023-06-14:
i tried that directly from cmd but not sure if its wrong
python.exe: No module named pylink.Jlink

if i start python and run the command its the same.. 
python
>> import pylink
>> jlink = pylink.Jlink()
same as tutorial

hkpeprah, 2023-06-14:
Can you try it lowercase `python -m pylink.jlink` not `python -m pylink.Jlink`? The code from the tutorial is:
```
>> import pylink
>> jlink = pylink.JLink()
```
I believe.

s-bruce13, 2023-06-14:
i tried both lowercase and upper case.

C:\Users\nxf51209\OneDrive - NXP\KW45\TRNG2>python -m pylink.jlink
C:\Users\nxf51209\AppData\Local\Programs\Python\Python310\lib\runpy.py:126: RuntimeWarning: 'pylink.jlink' found in sys.modules after import of package 'pylink', but prior to execution of 'pylink.jlink'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))

C:\Users\nxf51209\OneDrive - NXP\KW45\TRNG2>python -m pylink.Jlink
C:\Users\nxf51209\AppData\Local\Programs\Python\Python310\python.exe: No module named pylink.Jlink

When updating pylink is there any additional steps other than:  pip install pylink-square --upgrade?

hkpeprah, 2023-06-14:
This was what I was looking for. Seems like it imported fine without issue:
```
C:\Users\nxf51209\OneDrive - NXP\KW45\TRNG2>python -m pylink.jlink
C:\Users\nxf51209\AppData\Local\Programs\Python\Python310\lib\runpy.py:126: RuntimeWarning: 'pylink.jlink' found in sys.modules after import of package 'pylink', but prior to execution of 'pylink.jlink'; this may result in unpredictable behaviour
warn(RuntimeWarning(msg))
```
Are you still on `0.8.2`? Can you upgrade and try the above again?

s-bruce13, 2023-06-14:
ok i upgraded it again, if i run the commands i get the same result.

However i just tried running the script again and it is working. 

Dont know what changed, but thanks!

hkpeprah, 2023-06-14:
Good to hear!

closed 2023-06-14 by hkpeprah