square/pylink #183: Loading library on Aarch64 fails because pylink attempts to load 32-bit library

FletcherD, 2023-09-02:
The ARM64 version of the JLink software contains two library files, libjlinkarm.so (64 bit) and libjlinkarm_arm.so (32 bit). Pylink attempts to load libjlinkarm_arm.so first, which fails on ARM64 due to the architecture mismatch and causes an error. This is the error message:

`
>>> jlink = pylink.JLink(use_tmpcpy=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/odroid/.local/lib/python3.11/site-packages/pylink/jlink.py", line 293, in __init__
    lib = library.Library(use_tmpcpy=use_tmpcpy)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/odroid/.local/lib/python3.11/site-packages/pylink/library.py", line 293, in __init__
    self.load_default()
  File "/home/odroid/.local/lib/python3.11/site-packages/pylink/library.py", line 351, in load_default
    return self.load(path)
           ^^^^^^^^^^^^^^^
  File "/home/odroid/.local/lib/python3.11/site-packages/pylink/library.py", line 410, in load
    self._lib = ctypes.cdll.LoadLibrary(lib_path)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /opt/SEGGER/JLink_V792c/libjlinkarm_arm.so.7.92.3: wrong ELF class: ELFCLASS32
`

I have a proposed fix in https://github.com/square/pylink/pull/182 .

rviollette, 2023-12-11:
The same error occurs on Linux too (64-bit Debian-based system).
However, this fix only targets Windows.

In our case, the workaround is to instantiate `Library` class with `dllpath` parameter set to the full path of the `libjlinkarm.so` file.

FletcherD, 2023-12-11:
The fix is meant to target Linux and works for me, what issue are you having?

hkpeprah, 2024-07-22:
This is resolved by #182 right?