square/pylink #185: Error when programming or erasing Kinetis microcontrollers

Agarmonz, 2023-09-26:
Hi!

I'm trying to program devices with different types of microphones, for example: KE15, KL26, K64...
In this case all from the Kinetis family.

I find these two behaviours that I don't know how to solve:

- K64: it must be programmed starting from a memory that is already written. If you start from an erased device (for example with jFlash Lite), it fails.
- KE15: contrary to K64, I have to program it starting from an erased memory, if I do it starting from a programmed device, it fails.
- KL26: same as previous.

In all cases the error is as follows (mainly):
**_Failed to erase sectors 0 @ address 0x00000000 (Algo91: Flash protection violation. Flash is write-protected.)
Failed to erase sectors._**

Summarising the code steps I do the following:

1. jlink.open()
2. jlink.set_tif()
3. jlink.connect()
4. jlink.unlock()
5. jlink.halt()
6. jlink.erase()
7. jlink.unlock()
8. jlink.flash()

Unlock seems to work, so I have tried programming directly, without a previous erase. The failure is the same.

I have done something similar for other microcontroller families like LPC and iMX, and I have not had this problem, I can program and delete them without any problem.

Is it related to the library? Or, is it related to these kind of micros? Is a fault that has already been acknowledged and resolved?

Thank you in advance.
Hope you can help me asap :)

hkpeprah, 2023-09-26:
It sounds like you're possibly programming an image that protects the flash? You will need to erase unless you're writing to a region of flash that is already erased. Some sample code would help.

Agarmonz, 2023-09-27:
Hello again! 
I am not using images with security enabled. In fact, the idea is to program a very basic image and then put the necessary data in memory. And, in case this process is completed correctly, then write into memory the necessary data to activate the flash protection.

Example K64/KE15:
import pylink 
Import time 
Import sys

try: 
jlink = pylink.JLink() 
jlink.open() 
print(jlink.product_name) 
print(jlink.oem)
print('Open connection? %s' % jlink.opened()) 
print('JLink connected? %s' % jlink.connected()) 
print('Jlink found -> %s' % 
jlink.connected_emulators())
print('Connected device: %s' % jlink.target_connected())
except Exception as e:
print('ERROR: error when opening connection...', str(e)) jlink.close()
sys.exit() 
# ____________________________________________ #
 # ____________Target connection_______________ # 
# ____________________________________________ # 
try:
print('Specifying target interface...') jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
except Exception as e: 
print('ERROR: debugg...', str(e))
jlink.close()
sys.exit()

try:
print('Establishing the connection...') jlink.connect(microJLink, 'auto', True)
print(' -> Target connected')
except Exception as e:
print('ERROR: error when connecting, first attempt...', str(e))
jlink.close()
sys.exit()

 try:
print('Unlocking...')
pylink.unlockers.unlock(jlink, 'Kinetis')
print(' -> Unlocked')
except Exception as e:
print('ERROR: error when unlocking...', str(e)) jlink.close()
sys.exit()
# ____________________________________________ # 
# ________________Target erase________________ #
 # ____________________________________________ # jlink.halt()
print('CPU halted')
time.sleep(1)

try:
print('Erasing memory content...')
jlink.erase()
print(' -> Memory Erased')
except Exception as e:
print('ERROR: error when erasing...', str(e))
jlink.close()
sys.exit()

# ____________________________________________ #
 # ____________Target connection_______________ #
 # ____________________________________________ #
 try:
print('Establishing the connection...') jlink.connect(microJLink, 'auto', True)
print(' -> Connection established')
except Exception as e:
print('ERROR: error when connecting after erasing...', str(e))
jlink.close()
sys.exit()

try:
print('Unlocking...')
pylink.unlockers.unlock(jlink, 'Kinetis')
print(' -> Unlocked')
except Exception as e: print('ERROR: error when unlocking...', str(e))
jlink.close()
sys.exit() 
# ____________________________________________ # 
# ___________Flashing .s19___________ # 
# ____________________________________________ # time.sleep(1)
try:
print ('Programming module...')
jlink.flash_file(file_path, 0)
print(' -> Module programmed')
except Exception as e:
print('ERROR: error when programming...', str(e)) jlink.close()
sys.exit()

time.sleep(2)
# ____________________________________________ #
 # _______________Memory Write_________________ #
 # ____________________________________________ # 
try:
print('Memory write data1: %s (bytes)' % jlink.memory_write8(dirdata1, data1))

print('Memory write data2: %s (bytes)' % jlink.memory_write8(dirdata2, data2))

except Exception as e:
print('ERROR: error when modifying data...', str(e)) jlink.close()
sys.exit()
 # ____________________________________________ # 
# ___________Read Modified Memory_____________ #
 # ____________________________________________ # try: print('Reading memory...')
resultMemoryModifiedData1 = jlink.memory_read8(dirdata1, bytesdata1)

resultMemoryModifiedData2 = jlink.memory_read8(dirdata2, bytesdata2) 

except Exception as e:
print('ERROR: error when reading memory...', str(e)) jlink.close() 
sys.exit() 

# check mem
print('Checking memory...') 

if data1 != resultMemoryModifiedData1: print('ERROR: data1 FAILURE'); jlink.close(); sys.exit()

if data2 != resultMemoryModifiedData2: print('ERROR: data2 FAILURE'); jlink.close(); sys.exit()

print(' -> Memory checked')
 # ____________________________________________ # 
# _________________Security___________________ # 
# ____________________________________________ #
 try:
print('Memory write SEC: %s (bytes)' % jlink.memory_write8(dirSEC, SEC))
except Exception as e: 
print('ERROR: error when modifying SEC...', str(e)) jlink.close()
sys.exit() # ____________________________________________ # # _____________Cerrar la conexión_____________ # # ____________________________________________ # jlink.close()
print('OK: boot done')

As i said, this works on LPC and iMX family.

hkpeprah, 2023-10-04:
Hm. I'm having a hard time following. There is OTP that permanently locks flash on the Kinetis chips, so even if your image doesn't enable flash security, if you flashed one at one point that did, then flash access would be permanently locked. In terms of the code, I'm having a hard time following. Could you paste a block like:

```
jlink = pylink.Jlink()
jlink.connect(...)
<...>
```

Then point out the specific line in that block that is causing the issue and whether it is on the first time you run it or subsequent runs.