square/pylink #186: pylink.errors.JLinkException: Could not find supported CPU

enjeru15, 2023-10-12:
Hi,

I have encountered an issue. When I execute 'jlink.connect()' in my Python code, I receive an error message stating "Could not find supported CPU". Strangely, if I manually connect using JLink.exe or JFlashLite.exe BEFORE running my Python code, everything works as expected. The connection between the J-Link and the target device (nRF5340) is successful. Additionally, I have observed that if I plug and unplug the debug kit and then run my Python code, the connection becomes unsuccessful.

I suspect that there might be an initialization sequence missing in my Python code, which JLink.exe and JFlashLite.exe are able to handle automatically.

Here's a snippet of my source code. After running `jlinkConnection.connect(target_device, 'auto', verbose=True)` line, the error occurs.

`from pylink import jlink

jlinkConnection = jlink.JLink()

jlinkConnection.open (822001993)  # working - able to establish connection
target_device = 'nRF5340_xxAA_NET'

jlink_open_status = jlinkConnection.opened()
print(f"J-Link is open: {jlink_open_status}") # True when J-Link is open
jlink_connect_status = jlinkConnection.connected()
print(f"J-Link is connected: {jlink_connect_status}") # True when J-Link is connected

supported_target_interface = jlinkConnection.supported_tifs()
print(f"Supported target interfaces: {supported_target_interface}")  # 2 supported target interfaces
print(f"Current target interface: {jlinkConnection.tif}") # 0 - JTAG (default)
set_target_interface = jlinkConnection.set_tif(pylink.enums.JLinkInterfaces.SWD)  # Target interface is set to SWD
print(f"Jlink interface has changed: {set_target_interface}") # True when target interface has been changed

jlinkConnection.connect(target_device, 'auto', verbose=True)
target_device_connection_status = jlinkConnection.target_connected()
print(f"Target is connected: {target_device_connection_status}")`

Upon investigating the `connect()` module, the value of "result" changes:
`if not self.target_connected():
            result = self._dll.JLINKARM_Connect()
            if result < 0:
                raise errors.JLinkException(result)`

When I run Jlink.exe or JFlashLite.exe prior to running my code, the result = 0. However, when I unplug and re-plug the USB, result becomes -261.

enjeru15, 2023-10-13:
Hi, thank you for responding!
Here are the logs:

```
DEBUG:__main__:This is a debug message.
DEBUG:pylink.jlink:T0EE0 000:047.435   
DEBUG:pylink.jlink:Firmware: J-Link V12 compiled Aug  2 2023 10:34:27
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:048.909
DEBUG:pylink.jlink:Firmware: J-Link V12 compiled Aug  2 2023 10:34:27
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:073.978   
DEBUG:pylink.jlink:Hardware: V12.00
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:075.474
DEBUG:pylink.jlink:S/N: 822001993
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:076.669
DEBUG:pylink.jlink:OEM: SEGGER
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:077.354
DEBUG:pylink.jlink:Feature(s): GDB
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:079.582
DEBUG:pylink.jlink:Bootloader: 2022 Oct 19
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:080.554
DEBUG:pylink.jlink:USB speed mode: High speed (480 MBit/s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:081.254
DEBUG:pylink.jlink:TELNET listener socket opened on port 19021
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:082.019
DEBUG:pylink.jlink:WEBSRV WEBSRV_Init(): Starting webserver thread(s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:082.684
DEBUG:pylink.jlink:WEBSRV Webserver running on local port 19080
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T0EE0 000:083.151
DEBUG:pylink.jlink:Looking for J-Link GUI Server exe at: C:\Users\anramos\AppData\Local\Temp\1\JLinkGUIServer.exe

DEBUG:pylink.jlink:T0EE0 000:083.655
DEBUG:pylink.jlink:Looking for J-Link GUI Server exe at: C:\Users\anramos\SEGGER\JLink_V792g\JLinkGUIServer.exe

DEBUG:pylink.jlink:T0EE0 000:083.955
DEBUG:pylink.jlink:Forking J-Link GUI Server: C:\Users\anramos\SEGGER\JLink_V792g\JLinkGUIServer.exe

DEBUG:pylink.jlink:T0EE0 000:181.873   
DEBUG:pylink.jlink:J-Link GUI Server info: "J-Link GUI server V7.92g "

DEBUG:pylink.jlink:T0EE0 000:183.916
DEBUG:pylink.jlink:- 165.778ms returns "O.K."

DEBUG:pylink.jlink:T0EE0 000:184.355
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog

DEBUG:pylink.jlink:T0EE0 000:184.692
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog(...)
DEBUG:pylink.jlink:
[... 724 lines cut ...]
    return func(self, *args, **kwargs)
  File "C:\Users\anramos\AppData\Roaming\Python\Python39\site-packages\pylink\jlink.py", line 1138, in connect
    raise errors.JLinkException(result)
pylink.errors.JLinkException: Could not find supported CPU.
DEBUG:pylink.jlink:T0EE0 001:144.949
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T0EE0 001:145.979
DEBUG:pylink.jlink:- 1.029ms returns 0x01

DEBUG:pylink.jlink:T0EE0 001:146.842
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T0EE0 001:147.570
DEBUG:pylink.jlink:- 0.726ms returns TRUE

DEBUG:pylink.jlink:T0EE0 001:147.999
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T0EE0 001:148.458
DEBUG:pylink.jlink:- 0.458ms returns 0x01

DEBUG:pylink.jlink:T0EE0 001:158.549
DEBUG:pylink.jlink:JLINK_Close()
```

enjeru15, 2023-10-13:
And here's the log for the successful connection between J-Link and target device:

```
DEBUG:__main__:This is a debug message.
DEBUG:pylink.jlink:T652C 000:038.508
DEBUG:pylink.jlink:Firmware: J-Link OB-nRF5340-NordicSemi compiled Aug  8 2023 16:16:56
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:039.318
DEBUG:pylink.jlink:Firmware: J-Link OB-nRF5340-NordicSemi compiled Aug  8 2023 16:16:56
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:074.986
DEBUG:pylink.jlink:Hardware: V1.00
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:077.875
DEBUG:pylink.jlink:S/N: 1050028130
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:080.038
DEBUG:pylink.jlink:OEM: SEGGER
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:081.953
DEBUG:pylink.jlink:Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:084.655
DEBUG:pylink.jlink:Bootloader: 2021 May 18
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:086.592
DEBUG:pylink.jlink:USB speed mode: Full speed (12 MBit/s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:088.052
DEBUG:pylink.jlink:TELNET listener socket opened on port 19021
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:089.026
DEBUG:pylink.jlink:WEBSRV WEBSRV_Init(): Starting webserver thread(s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:089.868
DEBUG:pylink.jlink:WEBSRV Webserver running on local port 19080
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T652C 000:092.045
DEBUG:pylink.jlink:Looking for J-Link GUI Server exe at: C:\Users\anramos\AppData\Local\Temp\1\JLinkGUIServer.exe

DEBUG:pylink.jlink:T652C 000:092.717
DEBUG:pylink.jlink:Looking for J-Link GUI Server exe at: C:\Users\anramos\SEGGER\JLink_V792g\JLinkGUIServer.exe

DEBUG:pylink.jlink:T652C 000:093.150
DEBUG:pylink.jlink:Forking J-Link GUI Server: C:\Users\anramos\SEGGER\JLink_V792g\JLinkGUIServer.exe

DEBUG:pylink.jlink:T652C 000:199.802
DEBUG:pylink.jlink:J-Link GUI Server info: "J-Link GUI server V7.92g "

DEBUG:pylink.jlink:T652C 000:204.156
DEBUG:pylink.jlink:- 182.698ms returns "O.K."

DEBUG:pylink.jlink:T652C 000:204.924
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog

DEBUG:pylink.jlink:T652C 000:205.001
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog(...)
DEBUG:pylink.jlink:

[... 1203 lines cut ...]
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T652C 001:514.201
DEBUG:pylink.jlink:- 0.493ms returns 0x01

DEBUG:pylink.jlink:T652C 001:515.094
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T652C 001:515.374
DEBUG:pylink.jlink:- 0.279ms returns TRUE

DEBUG:pylink.jlink:T652C 001:515.731
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T652C 001:516.073
DEBUG:pylink.jlink:- 0.342ms returns 0x01

DEBUG:pylink.jlink:T652C 001:526.818
DEBUG:pylink.jlink:JLINK_Close()

DEBUG:pylink.jlink:T652C 001:527.624
DEBUG:pylink.jlink:CPU_ReadMem(4 bytes @ 0xE0001000)
DEBUG:pylink.jlink:

```

hkpeprah, 2023-10-16:
Could you try using the APIs around holding the [reset pin](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.set_reset_pin_high) or [reset strategy](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.set_reset_strategy)? It seems like the issue is that it's failing to connect under reset.

enjeru15, 2023-10-17:
Should I set the reset pin high or low? And when should the reset be triggered?

enjeru15, 2023-10-18:
Hi, I tried the following and it still "could not find supported CPU":
```python
jlinkConnection.set_reset_pin_low()
jlinkConnection.connect(target_device, 'auto', verbose=True)
``` 
```python
jlinkConnection.set_reset_pin_high()
jlinkConnection.connect(target_device, 'auto', verbose=True)
``` 
```python
jlinkConnection.set_reset_pin_low()
jlinkConnection.set_reset_pin_high()
jlinkConnection.connect(target_device, 'auto', verbose=True)
``` 
```python
jlinkConnection.set_reset_pin_high()
jlinkConnection.set_reset_pin_low()
jlinkConnection.connect(target_device, 'auto', verbose=True)
```

enjeru15, 2023-10-19:
Hi, I'm reviewing again the logs that I took.
Everything before the highlighted ones looked similar. They started to differ from the highlighted ones.

**Note**:
Unsuccessful connection was produced by unplugging/replugging the Jlink and the target device.
Successful connection was produced by plugging the Jlink and target device, launching the JFlashLite.exe, erasing the chip successfully using JFlashLite.exe, and then running my code.

[image]

enjeru15, 2023-10-19:
By the way, I'd like to mention that I'm currently using version V792g of the SEGGER J-Link Software package. Do you know if this version is 'compatible' with the pylink package? I'm exploring potential alternative causes for the issue I'm facing.

hkpeprah, 2023-10-19:
Looking at the above, I think the difference is that the flash EXE has extra logic in it to handle per-target differences, whereas this library is low-level and does not provide any target-specific handling. So if the NRF5340 (or the board in question) requires you to do something in order to hold the MCU in reset, you have to implement that logic yourself. Looking only, it seems like the nrf5340 is a dual core CPU. It is possible that the other core is interfering with your connection attempt?

enjeru15, 2023-10-19:
Yes, it seems the EXE file is running extra logic since the connection of J-Link and nRF5340 is always successful. And yes, nRF5340 is dual core ー application core and net core. Have you tried running the Pylink package with a dual core device? If so, did you run into the same issue?

Looking at the logs of both successful and unsuccessful connection, did you notice any differences? There seems to be no 'extra logic' before connecting to the target device.

[image]

enjeru15, 2023-10-19:
Are these the only cores supported by Pylink:

[image]
[image]

hkpeprah, 2023-10-19:
That is not an exhaustive list, no.

hkpeprah, 2023-10-19:
By extra logic, I mean that the EXE may be talking to the CoreSight interface, and using that to halt both cores, for instance. On some Cortex-M33s, the CoreSight interface has to be used to halt and/or erase the target.

enjeru15, 2023-10-20:
> By extra logic, I mean that the EXE may be talking to the CoreSight interface, and using that to halt both cores, for instance. On some Cortex-M33s, the CoreSight interface has to be used to halt and/or erase the target.

I see.. so it seems my code is lacking the `coresight_configure()` and `coresight_write()` commands before the `connect()` .
Upon looking at jlink.py, there is this function:
```python
@coresight_configuration_required
    def coresight_write(self, reg, data, ap=True):
        """Writes an Ap/DP register on a CoreSight DAP.

        Note:
          ``coresight_configure()`` must be called prior to calling this method.

        Args:
          self (JLink): the ``JLink`` instance
          reg (int): index of DP/AP register to write
          data (int): data to write
          ap (bool): ``True`` if writing to an Access Port register, otherwise
            ``False`` for Debug Port

        Returns:
          Number of repetitions needed until write request accepted.

        Raises:
          JLinkException: on hardware error
        """
        ap = 1 if ap else 0
        res = self._dll.JLINKARM_CORESIGHT_WriteAPDPReg(reg, ap, data)
        if res < 0:
            raise errors.JLinkException(res)
        return res
``` 
Where can I find the **list of integers (and their definitions)** that can be used for `reg` and `data`?
I think I may know now what the code is missing. It is not properly setting the core.
Based on this nRF5340 application note, the **AHB-AP must be set to 0** (for app core) **or 1** (for net core).

[image]

hkpeprah, 2023-10-20:
You'd have to bring up the reference manual for your chip. This is the [ARM Register List](https://developer.arm.com/documentation/100230/0004/appendices/debug-access-port/dap-register-summary/ahb-ap-register-summary), but for what values to right you would need the NRF's manual. There seems to be some documentation [here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf5340%2Fdebugandtrace.html) about how to properly connect.

[image]

enjeru15, 2023-10-24:
Thank you for this reference; it's very helpful! Although I still haven't figured out how to control the registers.

hkpeprah, 2023-10-25:
Here's the [`UICR`](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fuicr.html) register. The document I linked suggested that `APPROTECT` might be enabled and preventing debugger access: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf5340%2Fdebugandtrace.html

You can try and see by doing something like.
```
uicr = 0x10001000
app_reg = uicr + 0x208
jlink.coresight_configure()

app_protect = jlink.coresight_read(app_reg, ap=True) & 0xFF
if app_protect == 0xFF:
    print("Access Port Disabled")
elif app_protect == 0x5A:
    print("Access Port Disabled by Hardware")
elif app_protect == 0x00:
    print("Access Port Enabled")
else:
    print("Unknown Status")
```

I don't know if that code will work, but should give you an idea of how to use it. If it ends up being disabled, then you will have to enable it first before you attempt to connect the debugger. That can be done using the parallel API ][`coresight_write`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.coresight_write) according to what NRF's documentation says.

enjeru15, 2023-10-25:
H, I tried implementing this but using the specs of nRF5340 as this one is for nRF52832.
```python
uicr = 0x00FF8000
app_reg = uicr + 0x000
jlinkConnection.coresight_configure()

app_protect = jlinkConnection.coresight_read(app_reg, ap=True) & 0xFF
if app_protect == 0x00000000:
    print("Access Port Protected")
elif app_protect == 0x50FA50FA:
    print("Access Port Unprotected")
else:
    print("Unknown Status")
``` 
**It's printing "Unknown Status."**

nRF5340:
[image]
[image]

enjeru15, 2023-10-25:
I examined the `app_protect` value to understand why it's printing "unknown status." It currently shows as "0x43000002". Despite my efforts, I couldn't locate any relevant information regarding this value.

enjeru15, 2023-10-25:
I think this is the one I need to implement to resolve the issue - I need to disable the `approtect`:
[image]

It seems like  `approtect` is enabled every after reset.
However, as someone new in embedded, I don't fully understand how I should implement the disabling of `approtect` givent the information above. Could you advise me on how to do this?

hkpeprah, 2023-10-25:
Maybe this is the right documentation here? https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf5340%2Fctrl-ap.html&cp=3_0_0_7_9

The URL I gave you before was wrong; I was looking at the right thing, but it looks like the Nordic website doesn't use web history, so my URL wasn't updating to reflect the page I was looking at. Looks like `0x00` is `RESET` and not status, so I don't think that is the right one. It does look like `approtect` is enabled after reset, so you need to disable it over the CTRL-AP before attempting a connection. The reset value is `0x50FA50FA` on the debugger side, but looks like the CPU side is `0x00000000` on reset. It says the value of the field must match on both the CPU and debugger sides, and be non-zero for debug access.

It looks like the status is at `0x00C`, and disable is at `0x010`. I think we don't need the `UICR` address; I believe that is only relevant for the CPU side. I would try.
```
jlinkConnection.coresight_configure()

app_status_reg = 0x00C
app_status = jlinkConnection.coresight_read(app_status_reg, ap=True) & 0xFFFFFFFF
if app_protect == 0x00000000:
    print("Access Port Protected")
elif app_protect == 0x50FA50FA:
    print("Access Port Unprotected")
else:
    print("Unknown Status")
```
Then `0x010` is the register for the disabling protection on the debugger side. You will also need to read the `UICR.APPROTECT.STATUS` register on the CPU side to see if you have disabled it or not in firmware. If you have, then you will need to do a mass erase instead, and program an image that doesn't disable it in firmware (there's some details [here](https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/working-with-the-nrf52-series-improved-approtect)) for development. Provided the CPU doesn't disable, then it should work in theory.

enjeru15, 2023-11-13:
Hi,

Letting you know that I've already went to a different direction.
I dropped pylink and used a different Python package.

Thank you so much for your help!

closed 2024-07-20 by hkpeprah

ephraim271, 2024-09-26:
@enjeru15 What was the other python package?