square/pylink #188: Specify JLinkScript file

gw826943555, 2023-12-21:
Hi, 

I am now trying to connect a dual-core cortex-r5. We use `-JLinkScriptFile` command to specify the JLinkScript file and connect the core    
I'd like to connect. 
`JLink.exe -JLinkScriptFile xxx_path\R5_ConnectCore0.JLinkScript`
`JLink.exe -JLinkScriptFile xxx_path\R5_ConnectCore1.JLinkScript`
It works fine. 

My question is whether pylink support this feature.

Thanks!

hkpeprah, 2023-12-21:
PyLink does not support the scripts. The scripts are essentially commands / function calls. You will need to translate your scripts into the appropriate function calls.

gtowers-dukosi, 2024-08-09:
@hkpeprah -JLinkScriptFile does **not** specify a sequence of command function calls. You're getting confused with the -CommanderScript option.

JLinkScriptFile is a C-like syntax that allows you to customise internal actions, most notably what happens on a device reset to allow for hardware-specific requirements that cannot be easily dealt with through the standard J-Link API.
See https://wiki.segger.com/J-Link_script_files for more details.

I agree that supporting JLinkScriptFiles would be a really useful feature add for the module, as their lack limits what can be done with the module.

gtowers-dukosi, 2024-08-09:
The DLL is normally somewhere in _"C:\Program Files (x86)\"_, which on most machines needs admin privileges for an application to write to, so that's probably not going to work - but I like your thinking. Thanks for reading up, I'd missed that section on the page - very interesting to know that it's parsed by the .dll rather than the .exe. That suggests there must be another way to somehow specify a custom filepath directly to the DLL.

hkpeprah, 2024-08-09:
By default, we copy the library file to a separate temporary folder (unless you specify otherwise). So we should be able to write the `Default` script into that folder, and delete it when we clean-up the temporary library file as well.

gtowers-dukosi, 2024-08-09:
Sounds like it might just work then! I think it would meet my requirements in any case.
Thanks again

gtowers-dukosi, 2024-08-12:
I found a better solution! This does exactly what I need:
`jlink.exec_command("scriptfile = C:\\path\\to\\myJlinkScript.JLinkScript")`

Technically no pylink upfates even needed for pylink, but since it's not at all obvious I think a pylink wrapper function for that command would be a nice feature.

hkpeprah, 2024-08-12:
Nice! That is much simpler. If you want to submit a patch for the new function, I can accept it. Something like `set_script()`?

hkpeprah, 2024-09-04:
This is now available via `set_script_file()` in `v1.3.0` thanks to @gtowers-dukosi!

closed 2024-09-04 by hkpeprah