square/pylink #189: CORESIGHT_AddAP doesn't execut

engrnader, 2023-12-21:
Hi,
i'm executing a the commands below to connect to Cortex-M55 however I was not successful:
    response_1 = jlink.exec_command('CORESIGHT_AddAP = Index=4 Type=AHB-AP Addr=0x00300000')
    response_2 = jlink.exec_command('CORESIGHT_SetIndexAHBAPToUse = 4')

It seems that the second command works fine but not the first command. I wrote a jlink script and ran it from keil following the same exact procedure and I was able to connect to the device.

Can someone help me implementing this using pylink?

hkpeprah, 2023-12-21:
Could you try to use the `coresight_*` functions instead? https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.coresight_configure

engrnader, 2024-01-02:
If I use the coresight functions instead I would have to define the ir_pre and post and dr_pre and post. I don't know those parameters. 

How would you do it given the code I provided before?

hkpeprah, 2024-01-03:
AFAIK, you can omit those parameters (they're optional). What have you tried?

engrnader, 2024-01-03:
Then how can I specify the ROM table address?   Jlink by default assumes the ROM table is at address 0x0. 

I tried coresight_configure() with no parameters and it didn't work.

hkpeprah, 2024-01-08:
If there are multiple cores on the JTAG chain, then you unfortunately have to use the arguments: https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.coresight_configure

I personally haven't used an ARM device with multiple cores on a single scan chain, so I can't tell you what the arguments you would pass for each of those would be. Sorry to say. I would look up a reference manual for your part to see if it specifies anything about that. I'm guessing the `core_idx` is related to the index on the scan chain.

engrnader, 2024-01-09:
@hkpeprah thank you for your response. I will look into this more.