square/pylink #190: Can support C-JTAG

EasonNothingToSing, 2024-01-05:
Segger support c-jtag,can pylink implement it?

hkpeprah, 2024-01-08:
I'm not sure. Have you tried to connect and use it with your target? What is the output?

EasonNothingToSing, 2024-01-10:
In my chip that supports both C-JTAG and JTAG protocols, in pylink enum type, i can't  find corresponding C-JTAG type, so i just use set_tif(pylink.enums.JLinkInterfaces.JTAG) function?  i will try

hkpeprah, 2024-01-10:
I would try that.

EasonNothingToSing, 2024-01-19:
__jlink_dll = pylink.library.Library("JLink_x64.dll")

    jtag = pylink.JLink(lib=__jlink_dll)
    jtag.open()
    jtag.set_tif(pylink.enums.JLinkInterfaces.JTAG)
    jtag.connect(chip_name="N308")

    print(jtag.connected()) # Ture
    print(jtag.target_connected()) # False

I ran the above code and found that it can connect to J-Link, but the target cannot be connected. I'm not sure what the reason is

EasonNothingToSing, 2024-01-19:
[image]
I could connect N308 with jlink commander using C-JTAG

hkpeprah, 2024-02-01:
Maybe something naive, but could you try passing a different target interface from here for the target interfaces? 
Maybe `C2`?
https://github.com/square/pylink/blob/master/pylink/enums.py#L274

EasonNothingToSing, 2024-02-06:
i try it, but when i use C2, it will print "RISC-V is not supported via C2", and i then try other option, all trigger error

hkpeprah, 2024-02-06:
Are you sure your target is supported via C-2? This seems to suggest that C2 is proprietary to Silicon Labs MCUs: https://community.silabs.com/s/article/jtag-and-c2-debug-interfaces?language=en_US, which to my understanding are ARM-based.

hkpeprah, 2024-02-06:
Maybe you need to perform your own connect sequence for cJTAG, and use that instead of C2: https://wiki.segger.com/J-Link_cJTAG_specifics

hkpeprah, 2024-04-18:
Do you happen to know what error you saw?

EasonNothingToSing, 2024-04-28:
I found the macro for CJTAG in SEGGER, and by using version 7.9.4 of the DLL, calling my_dll.JLINKARM_TIF_Select(7) can successfully connect to CJTAG. Will this parameter be considered for inclusion in PyLink in the future?

hkpeprah, 2024-04-29:
Feel free to make a PR to add the enum value.