square/pylink #192: custom ResetTarget function in pylink

evertvr, 2024-02-05:
Is it possible to pass a custom ResetTarget function to pylink like is described on https://wiki.segger.com/J-Link_script_files#ResetTarget.28.29 ?

hkpeprah, 2024-02-05:
What's your code? If you run with logging verbose (`logging.basicConfig(level=logging.DEBUG)`), do you see the reset taking place?

evertvr, 2024-02-06:
Code would be:
```
from pylink import JLink
JLink.set_tif(1) # SWD
jlink.connect('CORTEX-M0', verbose=True)
```

Note that I don't touch the reset strategy in this case.

I've got the following debug trace (note when the devices comes out of reset, visible by some UART comms):

```
2024-02-06 08:27:44,715 DEBUG pylink.jlink: T2A1A4640 190:235.132
2024-02-06 08:27:44,716 DEBUG pylink.jlink: JLINK_IsOpen()

2024-02-06 08:27:44,716 DEBUG pylink.jlink: T2A1A4640 190:235.459
2024-02-06 08:27:44,716 DEBUG pylink.jlink: - 0.349ms returns 0x01

2024-02-06 08:27:44,716 DEBUG pylink.jlink: T2A1A4640 190:235.508
2024-02-06 08:27:44,716 DEBUG pylink.jlink: JLINK_IsOpen()

2024-02-06 08:27:44,716 DEBUG pylink.jlink: T2A1A4640 190:235.627
2024-02-06 08:27:44,716 DEBUG pylink.jlink: - 0.153ms returns 0x01

2024-02-06 08:27:44,716 DEBUG pylink.jlink: T2A1A4640 190:235.686
2024-02-06 08:27:44,716 DEBUG pylink.jlink: JLINK_EMU_IsConnected()

2024-02-06 08:27:44,716 DEBUG pylink.jlink: T2A1A4640 190:235.828
2024-02-06 08:27:44,716 DEBUG pylink.jlink: - 0.461ms returns TRUE

2024-02-06 08:27:44,717 DEBUG pylink.jlink: T2A1A4640 190:236.509
2024-02-06 08:27:44,717 DEBUG pylink.jlink: JLINK_ExecCommand("EnableRemarks = 1", ...).

2024-02-06 08:27:44,717 DEBUG pylink.jlink: T2A1A4640 190:236.612
2024-02-06 08:27:44,717 DEBUG pylink.jlink: - 0.023ms returns 0x00

2024-02-06 08:27:44,717 DEBUG pylink.jlink: T2A1A4640 190:236.673
2024-02-06 08:27:44,717 DEBUG pylink.jlink: JLINK_DEVICE_GetIndex(sDeviceName = CORTEX-M0)

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:241.686
2024-02-06 08:27:44,722 DEBUG pylink.jlink: - 5.125ms returns 16

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:241.888
2024-02-06 08:27:44,722 DEBUG pylink.jlink: JLINK_DEVICE_GetInfo(DeviceIndex = -1)

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:241.959
2024-02-06 08:27:44,722 DEBUG pylink.jlink: - 0.094ms returns 9824

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:242.068
2024-02-06 08:27:44,722 DEBUG pylink.jlink: JLINK_DEVICE_GetInfo(DeviceIndex = 16)

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:242.160
2024-02-06 08:27:44,722 DEBUG pylink.jlink: - 0.119ms returns 0

2024-02-06 08:27:44,722 DEBUG pylink.jlink: T2A1A4640 190:242.222
2024-02-06 08:27:44,722 DEBUG pylink.jlink: JLINK_ExecCommand("Device = CORTEX-M0", ...).

2024-02-06 08:27:44,723 INFO pylink.jlink: Device "CORTEX-M0" selected.
2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:242.373
2024-02-06 08:27:44,723 DEBUG pylink.jlink: Device "CORTEX-M0" selected.
2024-02-06 08:27:44,723 DEBUG pylink.jlink:
2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:242.609
2024-02-06 08:27:44,723 DEBUG pylink.jlink: - 0.384ms returns 0x00

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:242.751
2024-02-06 08:27:44,723 DEBUG pylink.jlink: JLINK_IsOpen()

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:242.863
2024-02-06 08:27:44,723 DEBUG pylink.jlink: - 0.144ms returns 0x01

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:242.930
2024-02-06 08:27:44,723 DEBUG pylink.jlink: JLINK_IsOpen()

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:243.043
2024-02-06 08:27:44,723 DEBUG pylink.jlink: - 0.165ms returns 0x01

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:243.145
2024-02-06 08:27:44,723 DEBUG pylink.jlink: JLINK_EMU_IsConnected()

2024-02-06 08:27:44,723 DEBUG pylink.jlink: T2A1A4640 190:243.245
2024-02-06 08:27:44,723 DEBUG pylink.jlink: - 0.144ms returns TRUE

2024-02-06 08:27:44,724 DEBUG pylink.jlink: T2A1A4640 190:243.356
2024-02-06 08:27:44,724 DEBUG pylink.jlink: JLINK_SetSpeed(0)

< DEVICE GOES OUT OF RESET AND BOOTS >

2024-02-06 08:27:45,276 DEBUG pylink.jlink: T2A1A4640 190:795.786
2024-02-06 08:27:45,276 DEBUG pylink.jlink: - 552.786ms

2024-02-06 08:27:45,277 DEBUG pylink.jlink: T2A1A4640 190:796.382
2024-02-06 08:27:45,277 DEBUG pylink.jlink: JLINK_IsOpen()

2024-02-06 08:27:45,277 DEBUG pylink.jlink: T2A1A4640 190:796.808
2024-02-06 08:27:45,277 DEBUG pylink.jlink: - 0.841ms returns 0x01

2024-02-06 08:27:45,278 DEBUG pylink.jlink: T2A1A4640 190:797.388
2024-02-06 08:27:45,278 DEBUG pylink.jlink: JLINK_EMU_IsConnected()

2024-02-06 08:27:45,278 DEBUG pylink.jlink: T2A1A4640 190:797.685
2024-02-06 08:27:45,278 DEBUG pylink.jlink: - 0.456ms returns TRUE

2024-02-06 08:27:45,278 DEBUG pylink.jlink: T2A1A4640 190:798.079
2024-02-06 08:27:45,278 DEBUG pylink.jlink: JLINK_IsConnected()

2024-02-06 08:27:45,279 DEBUG pylink.jlink: T2A1A4640 190:798.487
2024-02-06 08:27:45,279 DEBUG pylink.jlink: - 0.531ms returns FALSE

2024-02-06 08:27:45,279 DEBUG pylink.jlink: T2A1A4640 190:798.684
2024-02-06 08:27:45,279 DEBUG pylink.jlink: JLINK_Connect()

< DEVICE GOES OUT OF RESET AND BOOTS >

2024-02-06 08:27:45,800 DEBUG pylink.jlink: T2A1A4640 191:319.459
2024-02-06 08:27:45,800 DEBUG pylink.jlink: - 521.085ms returns 0xFFFFFFFF

< DEVICE GOES OUT OF RESET AND BOOTS >
```

running with pylink-square 1.1.0

evertvr, 2024-02-06:
connecting with a predefined speed `jlink.connect('CORTEX-M0', speed=1000,verbose=True)` seems to get rid of the first reset...

hkpeprah, 2024-02-06:
Instead of `Cortex-M0`, could you try the specific device name?

evertvr, 2024-02-06:
it's a custom device. Is there any device with cortex-M0 of which you know which has a "no reset" strategy which I could try? Or can I disable reset completely in pylink itself?

evertvr, 2024-02-09:
That seems to do the trick! Thanks.

closed 2024-02-09 by evertvr