square/pylink #193: Error "Failed to perform RAMCode-sided Prepare()" Fail "JLinkFlashException: Unspecified error."

jrigosa, 2024-02-20:
I have interfaced using pylink an msp432 board to robot framework in order to automate integration tests over CI.
Maybe out of scope, but on the CI the JLink is installed and configured on a container.

I have two tests, the first has 2 calls of `flash_file`: firmware, bootloader. 
Then the device boots and the test parses the log.
Test succeed. 

The second test has one more `flash_file` call for a config file, that is used in the test.
After every call the exception`Failed to perform RAMCode-sided Prepare()` is raised.
After last call the test fails due to  `JLinkFlashException: Unspecified error`. 
Curiously, I see twice the log `Percentage: 100` that would suggest the flash happened (firmware and bootloader ?). 
Curiously, the log `Percentage:` is printed only for the value 100, which I believe not to be the proper behavior (and in the first test, I could see two ramps from 0 to 100).

Is there any precaution I should consider to perform multiple subsequent Jlink flash calls?

hkpeprah, 2024-02-21:
Without a code sample, it's hard to say, but a precaution you have to take when attempting to program is that you have to ensure that the device is halted, and that nothing is interfering with the ability to toggle the reset line or hold the CPU in reset throughout the flashing process. A common example is a watchdog timer. If your software starts a watchdog timer, and you halt then program, the watchdog might fire during the programming attempt. In that case, you might want to halt, reset the CPU (to reset the watchdog), then program; or disable the watchdog first.

jrigosa, 2024-02-23:
Thank you very much for your reply.

closed 2024-04-03 by jrigosa