square/pylink #194: rtt_write only working when I have JLink RTT viewer open

carriegong, 2024-03-13:
I might have something set up wrong 
```
jlink = pylink.JLink()
print("connecting to JLink...")
jlink.open()
print("connecting to NRF52840_XXAA...") 
jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
jlink.connect("NRF52840_XXAA") # need to change this if we change the board we are working with
print("connected, starting RTT...")
jlink.rtt_start()
```

but my script works on write only when i have JLink RTT view open at the same time.

hkpeprah, 2024-03-13:
Can you run with verbose logging and see what the output is (when you don't have the RTT view open):
```
import logging

logging.basicConfig(level=logging.DEBUG)
```

carriegong, 2024-03-13:
> Can you run with verbose logging and see what the output is (when you don't have the RTT view open):
> 
> ```
> import logging
> 
> 
> logging.basicConfig(level=logging.DEBUG)
> ```

```
DEBUG:pylink.jlink:TB975E00 000:013.377
DEBUG:pylink.jlink:Firmware: J-Link V9 compiled May  7 2021 16:26:12
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:014.248
DEBUG:pylink.jlink:Firmware: J-Link V9 compiled May  7 2021 16:26:12
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:014.767
DEBUG:pylink.jlink:Decompressing FW timestamp took 286 us
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:029.333
DEBUG:pylink.jlink:Hardware: V9.60
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:029.609
DEBUG:pylink.jlink:S/N: 583650655
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:029.941
DEBUG:pylink.jlink:OEM: SEGGER
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:030.492
DEBUG:pylink.jlink:Feature(s): GDB, RDI, FlashBP, FlashDL, JFlash, RDDI
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:036.809
DEBUG:pylink.jlink:TELNET listener socket opened on port 19021
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:037.391
DEBUG:pylink.jlink:WEBSRV WEBSRV_Init(): Starting webserver thread(s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:037.927
DEBUG:pylink.jlink:WEBSRV Webserver running on local port 19080
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:038.156
DEBUG:pylink.jlink:- 35.340ms returns "O.K."

DEBUG:pylink.jlink:TB975E00 000:038.471
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog

DEBUG:pylink.jlink:TB975E00 000:038.644
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog(...)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TB975E00 000:038.930
DEBUG:pylink.jlink:- 0.517ms returns 0

connecting to NRF52840_XXAA...
DEBUG:pylink.jlink:TB975E00 000:039.060
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TB975E00 000:039.147
DEBUG:pylink.jlink:- 0.116ms returns 0x01

[... 412 lines cut ...]

DEBUG:pylink.jlink:TB975E00 000:273.579
DEBUG:pylink.jlink:- 0.223ms returns 0

DEBUG:pylink.jlink:TB975E00 000:277.974
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TB975E00 000:278.130
DEBUG:pylink.jlink:- 0.198ms returns 0x01

DEBUG:pylink.jlink:TB975E00 000:278.231
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:TB975E00 000:278.339
DEBUG:pylink.jlink:- 0.164ms returns TRUE

DEBUG:pylink.jlink:TB975E00 000:278.458
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TB975E00 000:278.576
DEBUG:pylink.jlink:- 0.173ms returns 0x01

DEBUG:pylink.jlink:TB975E00 000:286.205
DEBUG:pylink.jlink:JLINK_Close()
```

hkpeprah, 2024-03-13:
The log seems to be fine. Can you elaborate on what issue you're running into? Is it throwing an exception when the `RTT Viewer` isn't open, or?

carriegong, 2024-03-13:
> The log seems to be fine. Can you elaborate on what issue you're running into? Is it throwing an exception when the `RTT Viewer` isn't open, or?

No just not writing.. 
the return for rtt_write is 0 when the viewer is not open. But return is 5 when rtt viewer is open. Not sure why

hkpeprah, 2024-03-13:
Can you try running this example to see if it detects the UP and DOWN buffers? https://github.com/square/pylink/blob/master/examples/pylink-rtt

hkpeprah, 2024-03-14:
The read/write do not need to be on (different) threads.

carriegong, 2024-03-14:
well turns out i just needed to add a time.sleep() before writing.. odd. Thanks for your help :)

closed 2024-03-14 by carriegong