square/pylink #197: 1.2.0: Some tests are incompatible with Python 3.12

dvzrv, 2024-04-05:
Hi! :wave: 

I package this project for Arch Linux.
We are currently rebuilding against Python 3.12.

I noticed the following tests are now failing:

```
=========================== short test summary info ============================
FAILED tests/unit/test_jlink.py::TestJLink::test_cp15_register_write_success
FAILED tests/unit/test_jlink.py::TestJLink::test_jlink_restarted - AttributeE...
FAILED tests/unit/test_jlink.py::TestJLink::test_set_log_file_success - Attri...
FAILED tests/unit/test_library.py::TestLibrary::test_linux_dl_oserror - Attri...
FAILED tests/unit/test_library.py::TestLibrary::test_linux_dl_unavailable - A...
FAILED tests/unit/test_library.py::TestLibrary::test_linux_glibc_unavailable
======================== 6 failed, 436 passed in 1.13s =========================
```

[python-pylink-square-1.2.0-2-x86_64-build.log](https://github.com/square/pylink/files/14890625/python-pylink-square-1.2.0-2-x86_64-build.log)
[python-pylink-square-1.2.0-2-x86_64-check.log](https://github.com/square/pylink/files/14890624/python-pylink-square-1.2.0-2-x86_64-check.log)

FTR: I am also applying https://github.com/square/pylink/pull/150 as we have removed mock from the repositories.

dvzrv, 2024-04-05:
The test_library.py tests I can fix by replacing occurrences of `assertEquals` with `assertEqual`.

The other test failures seem harder to fix, maybe you have an idea?

I will ignore them for now

frogamic, 2024-07-09:
Hi @hkpeprah, I maintain a NixOS package that depends on pylink and ran into the same issue as @dvzrv. I have opened a PR that I think resolves this, note that I did find 2 tests that appear to be broken.

closed 2024-07-12 by hkpeprah

hkpeprah, 2024-07-12:
Should be fixed in `1.2.1`.

PR #204 (2024-07-12): Python 3.12 test updates
Renamed test function calls that are removed in Python 3.
Fixed some test cases that likely were broken but failure was masked by calling an invalid function.
Closes #197
commit 09898e7d (2024-07-12): [PR-204] Python 3.12 test updates (#204)

Renamed test function calls that are removed in Python 3. Fixed some test
cases that likely were broken but failure was masked by calling an invalid
function. Fixes #197.