square/pylink #200: REQ:  Please Add Support For:  Keil JLinkEFM8.dll

jwestmoreland, 2024-05-09:
Hello All,

I'm wondering if it's in the realm of possibility to add support for the Keil JLinkEFM8.dll - I suppose this could be a pull request for the future....

Just wanted to ask/bounce off the idea.

Thanks,
John W.

hkpeprah, 2024-05-22:
Sorry about the delayed response here. Can you clarify what this would be for? My understanding is that Keil is a compiler.

jwestmoreland, 2024-05-22:
Ford,

To support this:

https://www.silabs.com/development-tools/wireless/simplicity-link-debugger?tab=overview

Thanks,
John

hkpeprah, 2024-05-22:
Ah. I think your issue is the same as this one: https://github.com/square/pylink/issues/190
You need the C2 interface, not necessarily the EFM8 specific library file.