square/pylink #201: pylink.errors.JLinkException: Unspecified error - during connect()

darlo-concepcion, 2024-05-21:
Hi,

I've read the tips regarding [unspecified-error-during-connect](https://pylink.readthedocs.io/en/latest/troubleshooting.html#unspecified-error-during-connect). I confirmed that I was able to connect to the target with just JLink.exe. As shown below, I tried to dig a little further into pylink by manually executing some of the lines in JLink.connect(). I'm running Python 2.7.13 on a Win11 machine. I was trying to figure out whether there's a compatibility issue but wasn't sure given I couldn't call some functions.

Thanks

Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 20:42:59) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylink
>>> j = pylink.JLink()
>>> j.open(serial_no="<xxxxxxxx>")
>>> j.product_name
u'SEGGER J-Link'
>>> j.oem
>>> j.opened()
True
>>> j.connected()
True
>>> j.target_connected()
False
>>> j.connect("STM32U585CI")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pylink\jlink.py", line 142, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Python27\lib\site-packages\pylink\jlink.py", line 1138, in connect
    raise errors.JLinkException(result)
**pylink.errors.JLinkException: Unspecified error.**
>>> print(j.get_device_index("STM32U585CI"))
9533
>>> print(j.supported_device(9533))
STM32U585CI <Core Id. 235012351, Manu. ST>
>>> print(j.exec_command('Device = STM32U585CI'))
0
>>> print(j.set_speed(auto=True))
None
>>> print(j._dll.JLINKARM_Connect())
-1
>>> print(j.compile_date)
Apr 17 2024 16:21:58
>>> print(j.version)
7.96e
>>> print(j.compatible_firmware_version)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pylink\jlink.py", line 142, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Python27\lib\site-packages\pylink\jlink.py", line 1232, in compatible_firmware_version
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.
>>> print(j.firmware_version.split('compiled')[0])
J-Link V12
>>> print(j.error)
None

hkpeprah, 2024-05-21:
I noticed that you didn't call [`set_speed`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.set_speed) or [`set_tif`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.set_tif). Is it possible that you intend to connect to the target over SWD, and not JTAG? By default, the target interface will be JTAG, though I think most individuals are using SWD to connect to their ARM MCUs unless they have some other custom setup:
```
import pylink
j = pylink.JLink()
j.open(serial_no="")
j.set_speed(speed='auto')
j.set_tif(pylink.JLinkInterfaces.SWD)
j.connect("STM32U585CI")
```

darlo-concepcion, 2024-05-21:
Your set_tif() line made it work for me. I was under the mistaken impression that SWD was the default and you needed to do something special for JTAG.

Thanks!

closed 2024-05-21 by darlo-concepcion