square/pylink #209: Option to set RTT Search Range

j0ntraust1, 2024-08-22:
Hi,

would it be possible to add to the input of the rtt_start() a option to set the search range. Similarly to the option in the Rtt-viewer or the rtt command SetRTTSearchRanges . 

[image]

-- 

We are using chipset that doesn't work well with the auto detection and the placement in the ram is not always a exact number. We are currently doing workaround by having a while loop that searches for the block address but it seems way slower then it needs to be.

supersonic71, 2024-10-17:
> We are using chipset that doesn't work well with the auto detection and the placement in the ram is not always a exact number. We are currently doing workaround by having a while loop that searches for the block address but it seems way slower then it needs to be.

Here is a temporary workaround : You can use elftools in python to find out exactly what the rtt address is. This way you know where to connect to. 

```
def get_symbol_address(elf_file_path, symbol_name):
    try:
        with open(elf_file_path, 'rb') as f:
            elf = ELFFile(f)
            
            # Iterate through all sections in the ELF file
            for section in elf.iter_sections():
                if section['sh_type'] == 'SHT_SYMTAB':
                    # Iterate through all symbols in the section
                    for symbol in section.iter_symbols():
                        if symbol.name == symbol_name:
                            return symbol['st_value']
    except Exception as e:
        print(f"Error reading ELF file: {e}")
        return None

    print(f"Symbol '{symbol_name}' not found")
    return None

get_symbol_address("abc.elf", "_SEGGER_RTT")
```

j0ntraust1, 2025-04-30:
Thank you ! I did something similar, searching the .map file for SEGGER_RTT. 

[image]

hkpeprah, 2025-04-30:
Yes, we could implement this by reading the memory. Maybe as a separate method (e.g. `rtt_get_block_address()`). If someone wants to implement that, we would gladly accept it, otherwise we can look into it when time permits. The `.map` file is likely the fastest approach, but that would be done outside of `pylink`; for inside `pylink`, we would want to read the target memory to search for the symbol (magic string).