square/pylink #213: Feature request: specific exception for 'Could not find supported CPU' JLinkException

juhhov, 2024-10-09:
This is often symptom of blocked SWD access due to security reasons.
Would it be possible to have specific exception type for this error to not have to rely on the exception message.

hkpeprah, 2024-10-15:
It seems like there is a specific error code, so this seems easy enough to do: https://github.com/square/pylink/blob/master/pylink/enums.py#L66C32-L66C44

You're welcome to create a patch to add it in, otherwise I will try to get to it by EoW.

hkpeprah, 2024-10-23:
I started looking at this today, and I'm seeing that the result is `Unspecified Error` when this happens during `connect(https://pylink.readthedocs.io/en/latest/troubleshooting.html#unspecified-error-during-connect)`, which can mean different things (see [here]()). Was there a specific instance you were thinking of where it means SWD for sure?