square/pylink #214: [Feature request] Add support for power trace

diggit, 2024-11-09:
Hi,
I started using power output on my J-link base compact and noticed there is current and voltage available on it's webserver http://localhost:19080/target_power.htm .

[image]

[Segger website sites state only higher end probes support power tracing at 200 ksa/s](https://www.segger.com/products/debug-probes/j-link/technology/power-profiling/), but my J-link compacts can obviously measuring output current and voltage (at least on some lower rate, web control panel traces refresh several times per second.

There are API functions [JLINK_POWERTRACE_Control](https://github.com/square/pylink/blob/0b3e8ca8768725b27b1fd58df71f7110084ee6fb/pylink/library.py#L65) and [JLINK_POWERTRACE_Read](https://github.com/square/pylink/blob/0b3e8ca8768725b27b1fd58df71f7110084ee6fb/pylink/library.py#L66). It looks similar to RTT API functions.

Would it be possible to explore this functionality and add it into pylink? Unfortunately I don't have SDK license so I don't know if I can help.

hkpeprah, 2024-11-20:
Sorry, I forgot to update here. I'll take a look at this. I have the code complete, just haven't had a moment to test it yet. Will try to get something out by EoW.

hkpeprah, 2024-12-02:
Sorry, I'm just wrapping up the unit tests, then I will put up a patch.

hkpeprah, 2024-12-02:
PR is up! https://github.com/square/pylink/pull/215

hkpeprah, 2025-01-07:
This should be available in `1.4.0`.

closed 2025-01-24 by hkpeprah