square/pylink #217: Cannot unlock nrf52833 MCU

KWolfe81, 2024-12-18:
New nrf52833 rev B MCUs feature enhanced flash protections.  I cannot connect to these devices, getting the following error when I attempt to use jlink.connect(...)

> Traceback (most recent call last):
>   File "C:\Code\workspace\v2\test\mfg\test_controller.py", line 5, in <module>
>     import rtt as lib
>   File "C:\Code\workspace\v2\test\mfg\rtt.py", line 717, in <module>
>     program_firmware('C:\\Code\\workspace\\v2\\controller\\build\\zephyr\\merged.hex')
>   File "C:\Code\workspace\v2\test\mfg\rtt.py", line 40, in program_firmware
>     jlink.connect("NRF52833_xxAA", speed="auto", verbose=True)
>   File "C:\Python\Python312\Lib\site-packages\pylink\jlink.py", line 142, in wrapper
>     return func(self, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Python\Python312\Lib\site-packages\pylink\jlink.py", line 1138, in connect
>     raise errors.JLinkException(result)
> pylink.errors.JLinkException: Unspecified error.

If I connect using JLink.exe, I am prompted to run a flash recovery on the device, which works fine.  After that, I can connect using pylink fine up until I perform a flash erase.

Note: the nordic pynrfjprog library features a 'recover()' function which performs the same recovery as above (https://github.com/NordicSemiconductor/pynrfjprog/blob/master/pynrfjprog/HighLevel.py)

hkpeprah, 2024-12-18:
Unlocking methods have to be implemented per-chip based on the debug unlock method required by the vendor. I think this issue has been raised before, unfortunately I don't have an NRF target implement it against. This blog post seems to outline some details for doing it: https://limitedresults.com/2020/06/nrf52-debug-resurrection-approtect-bypass/

It looks relatively simple:
```
# Read APPROTECTSTATUS 
# (0x0 Access port protection enabled - 0x1 APP disabled) 
> nrf52.dap apreg 1 0x0c 0x00000000
# Write ERASEALL register
> nrf52.dap apreg 1 0x04 0x01
> reset
``

```
import enum
import time

class Regs(enum.IntEnum):
    RESET = 0x00
    ERASEALL = 0x04
    ERASEALLSTATUS = 0x08
    APPPROTECTSTATUS = 0x0C
    IDR = 0xFC

def unlock_nrf(dbg: pylink.JLink) -> bool:
    dbg.coresight_configure()
    status = dbg.coresight_read(reg=Regs.APPPROTECTSTATUS, ap=True)
    if status & 0x1 == 0x1:
        # Access port protection already disabled, so just return early.
        return True

    # Access port protection enabled, perform an erase.
    _ = dbg.coresight_write(reg=Regs.ERASEALL, 0x1, ap=True)

    start_time = time.time()
    while (time.time() - start_time < 15):
        time.sleep(0.1)
        status = dbg.coresight_read(reg=Regs.ERASEALLSTATUS, ap=True)
        if status == 0x00
            break

    if status != 0x00:
        # Unlock failed.
        return False

    # Reset the target via the CTRL-AP to complete the erase and unlock.
    _ = dbg.coresight_write(reg=Regs.RESET, 0x1, ap=True)
    _ = dbg.coresight_write(reg=Regs.RESET, 0x0, ap=True)

    # Clear the erase bit.
    _ = dbg.coresight_write(reg=Regs.ERASEALL, 0x0, ap=True)
    return True
```

Some more details here: https://docs.nordicsemi.com/bundle/nwp_027/page/WP/nwp_027/nWP_027_erasing.html.

hkpeprah, 2024-12-18:
Here's how that would be used:

```
import pylink
import pylink.unlockers.unlock_nrf as unlock_nrf

dbg = pylink.JLink()
dbg.open()
try:
    dbg.connect("...")
except pylink.JLinkException:
    if not unlock_nrf.unlock_nrf(dbg):
        raise RuntimeError("Failed to unlock device.")
    dbg.connect("...")
<...>
```

If you would like to test / clean it up and submit a patch, I would gladly accept it.