square/pylink #218: AttributeError: module 'collections' has no attribute 'MutableSet'

ManPython, 2025-01-24:
Python311 issue (from pylint) (2.6.0)
```
Scripts\pyreverse.exe\__main__.py", line 7, in <module>
  File "E:\Python311-venv\Lib\site-packages\pylint\__init__.py", line 54, in run_pyreverse
    from pylint.pyreverse.main import Run as PyreverseRun
  File "E:\Python311-venv\Lib\site-packages\pylint\pyreverse\main.py", line 14, in <module>
    from pylint.config.arguments_manager import _ArgumentsManager
  File "E:\Python311-venv\Lib\site-packages\pylint\config\arguments_manager.py", line 19, in <module>
    from pylint import utils
  File "E:\Python311-venv\Lib\site-packages\pylint\utils\__init__.py", line 10, in <module>
    from pylint.utils.docs import print_full_documentation
  File "E:\Python311-venv\Lib\site-packages\pylint\utils\docs.py", line 13, in <module>
    from pylint.utils.utils import get_rst_section, get_rst_title
  File "E:\Python311-venv\Lib\site-packages\pylint\utils\utils.py", line 8, in <module>
    import isort.api
  File "E:\Python311-venv\Lib\site-packages\isort\__init__.py", line 25, in <module>
    from . import settings  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python311-venv\Lib\site-packages\isort\settings.py", line 34, in <module>
    from .pie_slice import itemsview, lru_cache, native_str
  File "E:\Python311-venv\Lib\site-packages\isort\pie_slice.py", line 362, in <module>
    class OrderedSet(collections.MutableSet):
                     ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableSet'
```

hkpeprah, 2025-01-28:
Hello, can you list the steps that lead to this error being generated while working in this repo?

ManPython, 2025-01-28:
`pyreverse -o png -p main.py` - it's for pygame

hkpeprah, 2025-01-28:
This repository doesn't have a `main.py` or use `pyreverse`. Did you possibly mean to open this against the `pyreverse` repository?

ManPython, 2025-01-28:
I mind pygame as main base main.py script (it's MVC pygame).

ManPython, 2025-01-28:
Time ago I used this for django, and all was ok, but now want similar for pygame project - I updated all related around pyreverse and stocked.

hkpeprah, 2025-01-28:
This repository is `pylink`, did you mean to open this against `pylint`'s `pyreverse`? https://github.com/pylint-dev/pylint/tree/main/pylint/pyreverse

closed 2025-01-28 by hkpeprah

ManPython, 2025-01-28:
Soo.. wich repo is direct/orgnial for pyreverse regarding needs to create diagram as UML?

hkpeprah, 2025-01-28:
This repository is pylin**k**, you want the pylin**t** repository I linked above: https://github.com/pylint-dev/pylint/tree/main/pylint/pyreverse. `pyreverse` is a part of pylin**t**.

ManPython, 2025-01-28:
Ahh sorry for my mistake here ;]