square/pylink #221: jlink.restart() does not continue the processor execution - it remains halted

RobbieGalfrin, 2025-02-10:
Hello.
I'm using the pylink-square==1.4.0 with python 3.13.1 on windows.
I have JLink_Windows_V812d_x86_64.exe installed. Everything that is said below was also tested with JLink_Windows_V784e_x86_64.exe and was found to be identical. 

When I try to do jlink.halt()  and then jlink.restart(), , it remains halted and does not resume. I am connected to an arm-based chip "CYT2BL5CAA_M4" using swd, other functionality such as reading registers works. When I use Jlink commander the "Halt" and "Go" commands works as expected in the same setup.
`jlink.halted()
True
jlink.restart()
True
jlink.halted()
True`

what could be the issue? is it possible some specific dll version is required for proper operation?

hkpeprah, 2025-02-10:
Without looking at your sample code, it is a bit hard to tell, but if you're executing all three instructions in quick succession, it is possible that your chip is not yet running. You could try adding a small sleep between your call to `.restart()` and your `.halted()` check like `time.sleep(0.1)` (`100ms`).

RobbieGalfrin, 2025-02-10:
It's not a matter of delays. I tried also manually with enough delays between commands. 
Attached is the code.

What could possibly be the problem? 

`target_name = "CYT2BL5CAA_M4"
jlink = pylink.JLink()
jlink.open()
if not jlink.opened():
	raise ConnectionError("JLink failed to open")
jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
try:
	jlink.connect("CYT2BL5CAA_M4", speed='auto', verbose=False)
except pylink.errors.JLinkException as e:
	raise ConnectionError(f"JLink connection error: {e}")

if not jlink.target_connected():
	raise ConnectionError("JLink failed to connect")

jlink.reset()
time.sleep(1)
jlink.halt()
time.delay(1
if not jlink.halted():
	raise RuntimeError("JLink failed to halt")
jlink.restart()
if jlink.halted():  #returns True
	raise RuntimeError("JLink failed to restart")

`
[image]

hkpeprah, 2025-02-10:
Could you try:
```
jlink.reset()
time.sleep(1)
jlink.halt()
time.sleep(0.1)
if not jlink.halted():
    raise RuntimeError("JLink failed to halt")

jlink.restart()
time.sleep(0.1)
if jlink.halted():
    raise RuntimeError("JLink failed to restart")
```

RobbieGalfrin, 2025-02-11:
I dont understand why, but this one worked. 
Maybe it is the timing...
Thanks!