square/pylink #224: Flash and verify

boraozgen, 2025-03-03:
This is actually a question rather than an issue, but I don't know of any other platform to discuss this.

AFAIU it is not possible to control the flash write verification mechanism using the DLL. The SDK documentation mentions that `JLINKARM_EndDownload` and `JLINK_DownloadFile`, which underlie the `JLink.flash` and `JLink.flash_file` functions, return an error on verification failure, but that's it. So my question is, how do I control the verification process? Is it always on and not possible to turn off? Or does the configuration from the Control Panel as mentioned [here](https://kb.segger.com/J-Link_flash_programming) apply to the write operations from the DLL? I would be glad if someone can point me to the correct resources.

hkpeprah, 2025-03-24:
So sorry I missed this! I'm not sure there is such an option, but you could add a `try-catch` for the verification failure, and ignore it: https://pylink.readthedocs.io/en/latest/pylink.bindings.html#pylink.enums.JLinkFlashErrors.VERIFICATION_ERROR
```
try:
    ...
except errors.JLinkFlashException as err:
    if err.code != enums.JLinkFlashErrors.VERIFICATION_ERROR:
        raise err
    logger.error(f"Ignoring flash verification error: {err=}")
```

boraozgen, 2025-03-26:
Thanks for the reply. I don't actually want to turn off the verification, but make sure the verification happens instead. As there is no mechanism in the API to control this, I cannot make sure that the verification takes place.

michal-mlodecki, 2025-05-15:
@boraozgen , check out [JLink control panel](https://kb.segger.com/J-Link_Control_Panel).

I'm actually struggling with the same topic a bit, since I needed custom flash loader which initially did not implement L1 verification so flashing files supposedly failed, but the data was programmed, it was the verification that happened afterwards that was the source of error.

Later, it turned out, that by using control panel via html you can turn that off. Additionally you can turn off the L2 verification as well.
What's mor, you can store that exported file and load it to DLL from python - [command is here](https://kb.segger.com/J-Link_Command_Strings#ProjectFile).

Like that:
`
    result = jlink.exec_command(
        "ProjectFile = /path/to/control_panel_exported_file/exported_config_file.jlink"
    )`

However, not all of those configurations are exported to this file, there is no mention of L2 verification whatsoever.
Those L2 related options are probably hidden from the .jlink file and are not exported (at least for my case)
[image]

The only verification option I found within .jlink exported file is "VerifyDownload" or something similar.

Additionally, if you are interested, there is even a command that you can send to the DLL (similar to the example above) to skip L1 verification: [look here](https://kb.segger.com/J-Link_Command_Strings#SetSkipL1Verify), but there is none for L2 verification which is strange.

According to my current understanding of these things, only the L1 verification can be turned off. Already posted a question on Segger forum regarding the L2 verification and how to turn it off.