square/pylink #232: Help with adding more JTAG API functions

ksigurdsson, 2025-04-08:
Hi,

I'm trying to add support for the following JTAG API functions:

- `JLINKARM_JTAG_StoreInst()` - accesses TAP instruction registers
- `JLINKARM_JTAG_StoreData()` - accesses TAP data registers

I've used these functions successfully (via python) from an older version (v6) of the JLink software but something has changed and v8 now doesn't function quite the same way (I don't know if v7 worked the same).

I think what might have changed is the way one must express/setup/describe the JTAG chains to the JLink software. Previously you had to call void `JLINK_JTAG_ConfigDevices(U32 NumDevices, const JLINKARM_JTAG_DEVICE_CONF* paConf);` with a valid `JLINKARM_JTAG_DEVICE_CONF` structure that described the JTAG scan chain containing the following for each TAP in the chain:

```
- SizeofStruct (U32)
- IRLen (U32)
- IRPrint (U32)
- Id (U32)
- sName (const char*)
```

Do you happen to know if this is still all that's required or has the API changed?

Thanks

hkpeprah, 2025-04-28:
Hello, sorry for the long delay here. I can take a look. Unfortunately, the SDK is proprietary, so discussion or sharing of the documentation isn't allowed.

ksigurdsson, 2025-04-29:
No apology needed. If you're able to take a look at this sometime then I'd very much appreciate it. Thanks.