square/pylink #234: RTT no writing

maxd-nordic, 2025-04-28:
Hi!
I'm trying to replace the deprecated pynrfjprog with pylink to do RTT comms. I know that my setup *should* work because it does with that other library. Unfortunately, rtt_write always returns 0, even with added time.sleep() calls.
Reading seems to work fine, I get the boot banner of my app. Am I missing something?

Setup:
* pylink-square 1.4.0
* JLink V7.96i
* nRF5340 JLink OB (nRF9151-DK)

hkpeprah, 2025-04-28:
Hello, do you have sample code?

maxd-nordic, 2025-04-29:
slightly different setup, but same problem.
source: https://github.com/nRFCloud/utils/blob/f18a9cadfdab48c1e4dc4b237ae9edbf1f72d84b/src/nrfcloud_utils/comms.py#L376
log: [log.txt](https://github.com/user-attachments/files/19955620/log.txt)