square/pylink #237: Incorrect usage of return value in flash_file method

TheColector, 2025-05-23:
Hi,
I just found the following code inconsistency compared to what is stated both in DLL API and **flash_file** code documentation.

In flash_file documentation on return value, it says:
https://github.com/square/pylink/blob/9a06468a50fa9d91361e1795debbe2d91d2ff3fa/pylink/jlink.py#L2248-L2249

Whereas, in code, it retreives the number  of bytes written from DLL function called **JLINK_DownloadFile** - which actually does not return the number of bytes written, it returns either one of 6 status values (< 0 ) when error happens, or returns status ( > 0 ) when everything went OK
https://github.com/square/pylink/blob/9a06468a50fa9d91361e1795debbe2d91d2ff3fa/pylink/jlink.py#L2273-L2278

Unless I'm interpreting something wrong here, it might be some leftover from flash method implementation:
https://github.com/square/pylink/blob/9a06468a50fa9d91361e1795debbe2d91d2ff3fa/pylink/jlink.py#L2224-L2230

In this case, it uses different DLL API call and it returns the number of bytes written, so this is correct.
I can make a PR with fix for that, but wanted to ask first if it's really a bug or there is some other reason for behind that.

hkpeprah, 2025-06-09:
You're right. We should just return the `len(data)` on success. Would gladly welcome a PR to fix this!

TheColector, 2026-03-10:
Hi @hkpeprah
I've just published a fix for the issue: https://github.com/square/pylink/pull/263

Once it's through, we can close it.