square/pylink #238: OSError: exception: access violation when using x64 trying to use JLINKARM

waddahmoghram, 2025-07-09:
```
Exception ignored in: <function JLink.__del__ at 0x000001A0D5DDC680>
Traceback (most recent call last):
  File "...\AppData\Roaming\Python\Python313\site-packages\pylink\jlink.py", line 354, in __del__
    self._finalize()
  File "...\AppData\Roaming\Python\Python313\site-packages\pylink\jlink.py", line 412, in _finalize
    if self.connected():
  File "...\AppData\Roaming\Python\Python313\site-packages\pylink\jlink.py", line 459, in connected
    return self.opened() and bool(self._dll.JLINKARM_EMU_IsConnected())
  File "...\AppData\Roaming\Python\Python313\site-packages\pylink\jlink.py", line 448, in opened
    return bool(self._dll.JLINKARM_IsOpen())
OSError: exception: access violation writing 0x00007FFD6CEDFDD0
```

waddahmoghram, 2025-07-21:
It was a 64x platform for sure.

hkpeprah, 2025-07-24:
Do you have a sample script that reproduces the issue? Any details about the platform and environment you're running in?